FreeYourStream/freedeck-hardware #3: Create 3d files of the electronics to make it easier to design cases

koriwi, 2020-04-09:


hanswurstgmbh, 2020-04-12:
Someone already made the 3d files
https://grabcad.com/library/arduino-pro-micro-1
https://grabcad.com/library/display-oled-ssd1306-de-0-96-1

koriwi, 2020-07-02:
this issue is relevant again.
we have new (working) PCBs, so the 3d models need to be updated.

koriwi, 2020-07-17:
new pcb files and pcbs are here. i will assign this to myself

koriwi, 2020-08-10:
i'm doing some slight changes to the pcbs right now to avoid the buttons [short cuirciting](https://discordapp.com/channels/699366121593372752/699366419485163660/742492222569840801) the displays.
the pcbs dimensions also change very slightly, so existing 3d cases need to be enlarged @AdamWelchUK

jhaardt, 2021-02-09:
@koriwi
I saw you updated the trough-hole pcb to 1.1, is this the mentioned "slight changes"? If so, I would be willing to design a new case. But it takes some time, just ordered the parts this morning.

koriwi, 2021-02-09:
Yes.

We already have a new case. Soon to be published.
Feel free to design another one :)