FreeYourStream/freedeck-hardware #4: rough technical drawing

hanswurstgmbh, 2020-04-09:
It would be nice to have a rough sketch with the distances drawn on it. So a simple rough sketch similar to a technical drawing. So we could create a rough design for a housing in a cad software to be able to print it later directly in 3d.

Thanks!

koriwi, 2020-04-10:
https://github.com/koriwi/freedeck-hardware/blob/master/3x2/pcb/pdf/freedeck%203x2_etch_silk_top.pdf

koriwi, 2020-04-10:
https://github.com/koriwi/freedeck-hardware/blob/master/3x2/pcb/pdf/freedeck%203x2_etch_silk_bottom_mirror.pdf

koriwi, 2020-04-10:
these should be enough :)

closed 2020-04-10 by koriwi

koriwi, 2020-04-11:
https://github.com/koriwi/freedeck-hardware/pull/6

koriwi, 2020-04-11:
https://github.com/koriwi/freedeck-hardware/tree/master/3x2/3d/step

hanswurstgmbh, 2020-04-12:
i could not open the stp file and the pdfs do not have the distances, so it will be difficult to make something. 
But the case you designed is already very nice