FreeYourStream/freedeck-hardware #11: Device scaling

DarkLogicX, 2020-07-15:
I was wondering how large could this scale up to?  IE is it possible to have say 9 or more screen/buttons?

I was thinking it might be cool to have one as the max scale that it can go to, or at least as many as the StreamDeck XL.

And would it be possible for there to be a BOM and links to parts for such a scaled up version?

koriwi, 2020-07-17:
this is one for the FAQ
yes, it's possible right now to go up to 16 displays/buttons. arduino code is prepared for this
the configurator right now can only handle 3x2. so any help would be appreciated.

koriwi, 2020-07-17:
BOM, also one for the FAQ. We have the kicad files in the repo. you can zip them and import them to easy_eda where you can export a BOM and  a PickAndPlace file :)

closed 2020-07-17 by koriwi