FreeYourStream/freedeck-hardware #17: bigger version(s)

koriwi, 2020-08-10:
we are developing a concept for a 4x3 version (and similar) right now
We still need:
- through hole pcb files
- SMD pcb files
- 3d files
- cases designed around above mentioned 3d files

GitOeli, 2020-10-16:
Hello,
are you still looking for help on this one? I would really love to see a bigger Version of this awesome project! I have basic skills in PCB Design (mostly Eagle and KiCad, but I would be open to learn an new software in the process). Additionally I have experience with 3D Modelling in Solidworks, but I currently do not have access to this software. Maybe I can provide some help to make this one happen!

koriwi, 2020-10-18:
Hi. We just made a new PCB revision for the 3x2 version. If it proves to be a good design (including cases) you could base the 4x3 design on this new PCB and case (case follows soon)

GitOeli, 2020-10-19:
Hi, sounds good. Is there an ETA for the new PCB Design?
I played a bit around in easyEDA because it seemed to be the software that was used to create the PCB that is currently available. Seems to be the easiest PCB design software I've used so far. So if the new version is created in easyEDA aswell I could get started relatively quickly.
What Software are you using for the case 3D files?

koriwi, 2020-10-23:
The PCBs are on their way right now. i expect them to arrive latest in a few days. if i think it's good, i will update the pcb design in github. then you can easily start with a bigger version in easyEDA.

First case i wanted to design in fusion360, my default software, but recently i switched to openSCAD (because linux and so on)

GitOeli, 2020-10-25:
I am looking forward to the new design! How will the additional displays and switches handled on the arduino side? Two more signals (i.e. X3 and X4) and two more 4051s?

koriwi, 2020-11-01:
The new pcbs should arrive on monday!
For designs above 8 display i intend to use the widely and cheaply availabe 4067 16 channel multiplexer.

GitOeli, 2020-11-04:
Ah okay. So there will be only need for one more signal (D) between the ICs and the Arduino. 
That's good to hear. I am looking forward to getting started!

ZeddBlock, 2020-12-06:
Hi, very nice design, and awesome work ! Firstly, forgive my english (I'm french).
I plan to make this for some time now, but I would like a 4x3 or 5x3 or 4. 

My knowledges aren't enough to do it by myself. So I guess I'll wait your new one. 

Again thanks a lot for your work !

GitOeli, 2020-12-07:
I've created a PCB for a 4x3 Version, but I haven't tested it yet. I am not sure when I will be able to test this design and there isn't a case yet. 
[image]
[image]

ZeddBlock, 2020-12-07:
Ho, super nice !
But, for this design, you site to configure it won't work  ? And the arduino code would be very different ?

GitOeli, 2020-12-07:
As far as I can tell the arduino code is ready for more displays. Two minor changes in the settings.h file are needed and that's it. I don't know about the state of the configurator.

feedtechmobile, 2020-12-16:
Is it possible to update to receive colored icons? I'm looking for acrylic protectors like buttons for OLED screens for fear of compromising the OLED display @koriwi

koriwi, 2020-12-16:
Configurator needs to be updated for 4x3. It's not that much, but i have no bigger version for testing.

Color is not possible right now. Arduino has not enough processing power to refresh it fast enough

GitOeli, 2020-12-17:
I could provide the files for the 4x3 DIP Version if someone would want to test it. I will move at some uncertain point in the near future so I can't order stuff that will take some time to arrive. So I dont know when I will be able to test this design. But first I will post it in the discord to check for errors that I may have missed. I moved the additional signal for the multiplexers from 10 to 5 because 10 was already occupied with the SD Card and it was easier to route this way. I hope that doesn't cause any issues.