FreeYourStream/freedeck-hardware #23: Nothing displayed on screen

HarsiFPV, 2021-03-23:
Using all the components listed, I finished building it and when I plug it in after puting in the SD card containing the config file, the screens light up bright but do not display any image contained in the SD card. Using the serial function in the configurator doesn't help.
If you have any idea of what could be causing the issue feel free to tell me.
Thanks :)

Gregor-Davies, 2021-03-23:
Hey!
It would be helpful if you provided a picture of your wiring to make sure everything is connected properly.

HarsiFPV, 2021-03-23:
Hey ! My bad for not saying this but I am using the pcb that Adam Welch provides in his video. [
[image]
[image]

](url)

Gregor-Davies, 2021-03-24:
Did/does the Arduino work when not attached to the pcb? The pins on the Arduino might need a bit more solder.

HarsiFPV, 2021-03-24:
Yes I flashed the arduino with the sketch before soldering it. I will try to add a little bit more solder. I'll keep you in touch thanks :)

HarsiFPV, 2021-03-24:
So I wen't ahead and reflowed solder on every pin of the board, and I am still facing the same issue :(

Gregor-Davies, 2021-03-24:
In that case there's something wrong in the pcb try going around on the pcb with a multimeter.

legomannetje, 2021-04-20:
Hey HarsiFPV,

If you still have this issue, please visit our discord! [Click here to join](https://discord.gg/sEt2Rrd)

koriwi, 2021-05-19:
Feel free to reopen or comment your solution. will close this ticket now :)

closed 2021-05-19 by koriwi